Guzunty/Pi #31: Please add the PCB traces for testing/debuging the soldering job.

DrnSln, 2015-01-21:
I recently bought two Guzunty kits and soldered the components to the boards. But when testing the soldering job, one of the boards fails the "*Check between Ground and 3.3v. This should not be open circuit, but should show a very large value, upwards of 5 MegaOhm. If you hold the measurement for a while, you may see the reading increase.*" step. As I want to see where the solder is failing to make a proper contact, I would need the pcb traces. I installed Kicad and got the design files, but when starting PcbNew, I get the "*Board' ..\guzunty.kicad_pcb' does not exists.*"

campbellsan, 2015-01-22:
Hi, sorry to hear you're having trouble. The traces probably wouldn't help much. Can you post top and bottom images of your board? Depending on where you live, you may be able to send me the board and I can debug it for you.

campbellsan, 2015-02-01:
Hi added a trouble shooting page with a detailed board image, labeled with all the circuit nets. It should prove helpful for anyone having trouble with shorts or open circuits.

@DrnSln, note the suggestion to pull the VCCIO power jumper, this should help you narrow down the problem area considerably.

Closing this issue, please raise a new one if you have more questions.

closed 2015-02-01 by campbellsan

DrnSln, 2015-02-11:
Excellent, thank you very much for adding the board image, my issue is now solved ( a bad joint on the ground pin - more heat was needed maybe due to larger surface area, plus I used the [Stacking Header for Raspberry Pi](http://www.adafruit.com/products/1112) and soldered it first). 

Now, because this is too fun to play with, I would need two LCD daughterboards and one Bare PCB shipped to Canada - 3 * GBP 2.60 + GBP 3.30 p&p = GBP 11.1 - is it ok to buy the Complete Kit UK Only GBP 10.00 + GBP 2.20 p&p and put the PCBs list as a note  ?

campbellsan, 2015-02-11:
I'm really pleased you found the new page useful, and that you are now enjoying the world of programmable logic arrays. Pretty cool aren't they?

By all means use the payment method you described if you feel more comfortable, but you don't have to use the dedicated Guzunty ordering page. You can just make a PayPal payment for the exact amount if you like. Just go to your PP Account home page and click on 'Send Money'. The email address to use is guzunty (at) gmail (dot) com. You get the same payment security as if you'd used the Guzunty website.

HTH,

Derek