Guzunty/Pi #33: new core request - frequency divider / variable audio wordclock

johhar, 2015-02-09:
I have been wondering if the guzunty would be suitable for creating custom rate audio wordclock generator. Stepped between 10khz to 100khz is what I would like to achieve, but there would be other uses. I believe this is a 50% square wave.

The use for audio recording is discussed here:
http://www.sawstudiouser.com/forums/showthread.php?t=4675

campbellsan, 2015-02-09:
There is a related request #11, though that is merging actual audio streams as opposed to an audio  word clock.

The answer is that the CPLD is very good at dividing down at powers of 2. Finer tuning than that is possible, but starts to use more and more resources.

Have you looked that the control over the GP_CLK hardware clock on the Pi itself? I believe it uses clock cycle dropping to achieve a fine grain effect. The source clock is high frequency to keep the distortion caused by dropped clock cycles out of the audio range.

Perhaps a combination of that plus the CPLD would meet the need?

johhar, 2015-02-10:
I just found http://abyz.co.uk/rpi/pigpio/examples.html#Misc_code
which illustrates setting the clock. So yea, it looks pretty straightforward to use a combined approach.

How about using 4 or 5 gpio pins to describe an exponential divisor for the clock. For instance, with a 5 bit divisor starting with (125Mhz)/(2*(2^31)) gets a cycle time of ~half a minute - and that is before slowing down the clock. It could be years. A four bit divisor exponent would more than cover my need. 

Another possibility would be to have a multiple output pins clock/2, clock/4, clock/8... I don't know how useful this would be, but it might be handy for somebody.

campbellsan, 2015-02-10:
"which illustrates setting the clock". Yes MASH mode is what your application requires I believe. You can read some more details about it in the BCM2835 ARM Peripherals manual, p.105. It provides a table of example divisors and resulting output frequencies.

"Another possibility would be to have a multiple output pins" Check out the gz_test core. It already does exactly this.

johhar, 2015-02-15:
Thanks for the info...

I'm continuing this for the record - but it is looking less like a Guzunty issue. Word clock specification page 14:
www.aes.org/standards/comments/drafts/aes11-2009-cfc-reaff-141124.pdf

For what I would like to accomplish which is record acoustic sources at shifted sample rates so the played-back audio will be at a different pitch, I think integer division with no fractional component (so as to avoid MASH) will give me the least jitter (or most uniform pulses) and while dividing 125Mhz will not give me exactly 44.1khz or other 'official' sample rates, if you want to use them, integer division will get you quite close and be out of pitch less than one cent (one one-hundredth of a semitone) which is inaudible. For the record, the official sample rates are about being able to sync to video - which I'm not doing.

If anybody has a suggestion how to accurately echo the clock output so as to protect the pi (or if there is already an expansion board that does this), recommendations would be great, but it probably doesn't warrant a custom core.

johhar, 2015-02-16:
I might do this myself as a core anyway. It seems like it would be a beginner project. I'll take a crack at loading the ISE and see if I can get it to install on my machine. I skimmed the instruction months ago and thinking it was for windows. I took another look and see it installs on Linux (depending).