Guzunty/Pi #37: gz_load fails after first load of gz_test

lmuray, 2015-02-23:
I built the kit without problem, installed all the software, and ran through the test programs with gz_test.  However, now when I try to load a different core I get the following error...

pi@piToy ~/Guzunty/Pi/src/gz_16o8i $ sudo gz_load gz_16o8i.xsvf
Guzunty loader v5.01 rev 2, portions courtesy Xilinx, Inc.
XSVF file = gz_16o8i.xsvf
Progress indicator: |-------------------|
ERROR:  TDO mismatch and exceeded max retries
ERROR at or near XSVF command #48.  See line #48 in the XSVF ASCII file.

ERROR:  TDO mismatch and exceeded max retries
Execution Time = 0.090 seconds

The soldering looks ok... any suggestions?
Thanks!

campbellsan, 2015-02-23:
The most common cause of this error is the Guzunty board not being seated on the GPIO correctly. Next possibility is that the programming jumpers are not positioned correctly.

Also make sure you have the latest copies of the software. There have been a lot of updates in the last couple of weeks to track changes in the Raspian kernel.

If none of those is the problem please let us know. We've had a very few CPLDs which seem to program only once, in which case we'll get another out to you asap.

lmuray, 2015-02-23:
Thanks for the quick response....
1) I tried reseating the board several times - no success
2) The GZ software was downloaded and installed yesterday - the Pi software is NOOBS_v1_3_11.zip
3) Maybe we need to try swapping.

campbellsan, 2015-02-23:
Sorry to hear you're having a bit of trouble.

Please mail us at guzunty at gmail and we'll get you sorted out.

campbellsan, 2015-02-25:
Replacement CPLD on its way to you. Since we're not aware of any defects with the materials on this site at this time, I'm going to close this issue as addressed by the replacement.

When it arrives and if you still have issues, please feel free to reopen it.

closed 2015-02-25 by campbellsan

lmuray, 2015-02-25:
Thanks!  I really appreciate your great support!

JMishou, 2015-05-18:
Hi,

I have the same problem.  gz_test loaded and the test program worked perfectly on the first run but since I cannot change cores.

 Did the replacement CPLD fix your issue Imuray?

```
pi@raspberrypi ~/Guzunty/Pi/src/gz_lcd $ sudo gz_load gz_lcd.xsvf
Guzunty loader v5.01 rev 2, portions courtesy Xilinx, Inc.
XSVF file = gz_lcd.xsvf
Progress indicator: |-------------------|
ERROR:  TDO mismatch and exceeded max retries
ERROR at or near XSVF command #48.  See line #48 in the XSVF ASCII file.

ERROR:  TDO mismatch and exceeded max retries
Execution Time = 0.120 seconds
```

lmuray, 2015-05-19:
Hi – yes the replacement fixed it.  It is very strange. There are no issues now.

Larry