Guzunty/Pi #38: gz_load fails after first load of gz_test

JMishou, 2015-05-18:
I have the same problem as issue #37 

The CPLD programmed fine the first time using the gz_test core but now fails every time.  The qz_test program also no longer works.

```
pi@raspberrypi ~/Guzunty/Pi/src/gz_lcd $ sudo gz_load gz_lcd.xsvf
Guzunty loader v5.01 rev 2, portions courtesy Xilinx, Inc.
XSVF file = gz_lcd.xsvf
Progress indicator: |-------------------|
ERROR:  TDO mismatch and exceeded max retries
ERROR at or near XSVF command #48.  See line #48 in the XSVF ASCII file.

ERROR:  TDO mismatch and exceeded max retries
Execution Time = 0.120 seconds
```

campbellsan, 2015-05-18:
Hi JMishou,

Yes, it does sound like the same issue. Don't worry, we'll get you sorted out.

Did you order the kit from the website here or on eBay?

(BTW, lmurray did confirm to me that the issue was resolved with a replacement part)

JMishou, 2015-05-18:
Sorry for adding an additional issue for one that is already solved but I was not sure if the reply to the closed issue was going to notify anyone and I could not figure out how to message directly.

I purchased it on ebay from cabin-globa in North Carolina

http://www.ebay.com/itm/331556504079

campbellsan, 2015-05-18:
No problem on the issue, it's what they're here for!

I'll ask Mark MacMillan to send you a replacement CPLD. That way, you'll get it faster than you would if it was coming from the UK. If you would, please send me by private mail (guzunty at gmail) your postal address and I'll forward it to Mark.

Finally, please let me know here when you're back up and running and I'll close this issue. 

best,

Derek

JMishou, 2015-05-23:
Strange... I have the same issue with the new chip.  I was waiting for the replacement before sending back the old one.  I'll have both chips in the mail by the end of the weekend.  Is there anything I you would like me to try before sending them back?

campbellsan, 2015-05-24:
Hi JMishou, sorry you're having a bit of trouble.

I'll send you a CPLD from a different batch to see if that is the trouble, but 1) a faulty part is starting to look less likely and 2) it will take a bit longer to arrive, so let's look at alternative explanations first.

If you moved the jumpers to the run position, make sure they are back in the program position. Actually, those jumpers only ever need to be moved if your project requires access to more signals than the default configuration allows for.  

If you have any doubts at all about dry joints etc, please take high resolution photos of component and solder side and mail them to me. You never know, I may be able to see something in the images that would help. You could also try reflowing the solder on the relevant pins. I'll be happy to tell you which ones to focus on if you need guidance.

I had one thing to try in my TO DO list which may be relevant. That was to reduce the values of the JTAG pull-up resistors to 4k7 (they are currently 47k). I found this recently in a Xilinx application note, however I highly doubt this will make any difference because I have seen other CPLD breakout boards which function perfectly well with no pull-ups at all.

If none of these suggestions is relevant, I'll send you a CPLD from a different batch as I suggested.

JMishou, 2015-05-25:
Good evening.

I have attached a couple of photos of the solder joints.  They look good to me.  Please let me know which specific programming joints I should reflow and point out any that you think may be faulty.  What is odd though is that it programs fine the first time.  What is it that makes the first program different from subsequent programs?  I assume all the same pins are used.

I have not adjusted any of the jumpers.

[image]
[image]

campbellsan, 2015-05-25:
Hi JMishou,

Looks like you did a great job, so I've no reason to think you have dry joints.

If you do want to try reflowing, this link should help a lot:

https://github.com/Guzunty/Pi/wiki/Troubleshooting

You need to look for JTAG_TMS, JTAG_TDO, JTAG_TCK, JTAG_TDI and all the nets associated with those signals, GPIO23, GPIO17, GPIO24 and GPIO22.

I'll put a new CPLD in the mail to you tomorrow.

I'm prepared to send you a prebuilt and tested board if that is what it takes to get you going, so no worries. :-)

campbellsan, 2015-05-25:
You are right about the same pins being used on subsequent program cycles. It's not unknown for builds to fail after the first cycle, but I'll readily admit that it is very rare.

However, we have to work with the data we have. It worked once and now it doesn't.

There'll be a new CPLD in the mail tomorrow.

JMishou, 2015-05-29:
Sorry for the delay but both of the chips that failed on me are in the mail headed to Mark.  Please attempt to program them and report back to me.

JMishou, 2016-02-17:
Good evening... Life happened and my hobbies were put on hold but I just got around to the newest chip that was sent to me and it was successfully programmed and then reprogrammed.  It seems like I really did get two faulty chips.

Thank you for your support.

closed 2016-02-17 by JMishou