Guzunty/Pi #40: bcm2835-v4l2 not working with Raspbian 7.8, Python 2.7.3

campbellsan, 2015-07-12:
When the PiTeR code opens the camera, the following errors are reported:

libv4l2: error setting pixformat: Device or Resource busy
HIGHGUI ERROR: libv4l unable to ioctl S_FMT
HIGHGUI ERROR: V4L: Initial Capture Error: Unable to load initial memory buffer

https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=71320

Would seem to be related to this.

antrapra, 2015-07-15:
Replacing the SURF part.

When the program has not well finished, the v4l2 damaged, Causing not run the program later with my corrects. After reboot a couple of times finally i correct.

I had a little mistakes with threading, so I went out error v4l2. I could not find the error, because, when the program has not well finished, v4l2 could damage, Causing not run the program later with the correct code. After I reboot a couple of times, finally i correct the error.

The program runs very slow using the raspberry pi 1  model b, it has 3-4 second delay. And Without threading, only approximately 1 second delay.

campbellsan, 2015-07-16:
So your program error out and did not close the camera input? Yes, I think I have seen such problems during development. You can try to avoid leaving the camera locked by using a python 'finally:' clause which ensures the camera is released.

Are you saying your own program is running very slowly or the standard one published here?

It has been a while since I made the multi-threaded performance measurements, which were published here:

https://www.raspberrypi.org/forums/viewtopic.php?f=37&t=91269&p=789493&hilit=piter#p789493

but that was back in February. My original measurements did show that a single core Model B ran the multi-threaded version slightly slower. However, the slowdown was only 5-10% as I recall, not 3-4 times slower back then.

It is possible that later releases of Raspbian or OpenCV has made it run slower. I will try to make some time to repeat the measurements with the latest updates in place, but I can't promise when I will be able to get to it, sorry.

campbellsan, 2015-07-17:
Having set up all the hardware, I'm not reproducing your reported faults.

> i'am using raspberry pi model B, with raspbian 7.8, Python 2.7.3, bcm2835-v4l2

Given my inability to reproduce, to help any further, I need:

- Output of uname -a
- Correct version of OpenCV (2.4.1 is old and unsupported?, 2.4.10, 2.4.11?)
- Output of v4l2-ctl --info

antrapra, 2015-07-23:
I have v4l2 version 4.0.8, and i already install opencv2.4.10 but in virtual enviroment, and i'm taking problems with that. Exist other form to install opencv 2.4.10 on raspbian?. Or another new version

I fallow this guy: http://www.pyimagesearch.com/2015/02/23/install-opencv-and-python-on-your-raspberry-pi-2-and-b/#comment-338616

campbellsan, 2015-07-26:
Sorry, I don't know of any other way.

I'm going to take a look at upgrading the code to use OpenCV 3. However it may take me some time since I have a new day job that involves a lot of travelling.