Guzunty/Pi #41: Request - SBUS input

0xCCD, 2015-11-07:
Hi, i dont know if it is possible, but i have to ask for it :) since you managed PPM is there a chance Guzunty can handle *SBUS* as well ? 

I'm not sure what you need to know, but here are some resources Here a few resources:
Description: http://www.futabarc.com/sbus/
Comparison: http://blog.oscarliang.net/ppm-sbus-d4r-ii-x4r-sb/
Schemata for conversion: http://www.rcgroups.com/forums/showthread.php?t=1459431&page=2
Ready to use Decoder: http://www.hobbyking.com/hobbyking/store/__52185__FrSKY_SBUS_to_CPPM_Decoder.html

Looking forward to hear from you

campbellsan, 2015-11-07:
Looking at the protocol, it isn't possible without additional components and even then you'd be better off with a microcontroller based solution I think. I'd recommend looking at this thread if you didn't find it already: http://forum.arduino.cc/index.php/topic,99708.0.html

If you are interested in a single board solution that incorporates an arduino to mount on a Pi, I may be able to help you with that .... :-)

campbellsan, 2015-11-07:
BTW, the reason I'm so sure it can't be done is that CPPM is a stream of pulse widths which just need to be de-multiplexed to the individual servos. With SBUS, the servo positions are being transmitted as 16 digitally encoded values. A conversion back into a pulse width format would require cacheing the sixteen channel values and reproducing the PWM servo signal.

This something that a Guzunty can do, but not for 16 channels simultaneously. Cacheing the channel values alone would require 16 x 11 register bits and the CPLD has only 72, it might be able to decode the first 4 channels but that is not what you had in mind, I think.

Hence my recommendation to use a microcontroller; it may not have a response time measured in nanoseconds, but it has loads of memory to play with.

0xCCD, 2015-11-09:
Hi and sorry for my belated answer, my main intension is the navio+ board, which 'currently' does not support SBUS, so my main idea was then to use the guzunty on top of it (i'm not even sure those are stackable). 
For now, i've switched to a new Firmware with CPPM Support and ordered a Guzunty Board for playing a little bit. Basically i need just a few channels to trigger an action on the pi while flying like record a video or take a picture and such stuff. 

But i really like the idea to make a single board solution including an arduino, how complicated is something like this and is such a solution fast enough for Quad Copter ?

campbellsan, 2015-11-09:
I have a very few prototype boards which already do this. I've successfully flown a tri-copter with Multi-Wii running in it (though I just used the Multi-Wii CPPM rather than the Guzunty solution, you could use either).

I'll send you one of the prototype PCBs with your Guzunty kit and you can decide for yourself if you want to use it. You'll need to source some extra parts of course.

0xCCD, 2015-11-09:
sounds great ! 
Many thanks in advance