Guzunty/Pi #42: Order status

stevenverver, 2015-12-15:
Dear campbellsan,  I'm sorry to put this here but I can 't find your e-mail address anywhere or any other contact info.  Maybe wise to put your e-mail on the order page? 

My question: I ordered and pand a Guzunty on dec 5th but the package isn' t delivered yet and I didn' t get any order confirmation or status update yet. So: did you receive my order in good order? 

Greetings, 

Steven Verver

campbellsan, 2015-12-15:
Hi Steven,

One way to get in touch is as good as another. :-)

I checked my records and your Guzunty was shipped late on the 7th. I would have expected that you would have received it by now.

Perhaps the dreadful events of the past few weeks have slowed the European mail system?

If you would be so kind as to wait until Friday morning? If you have not received it by then I will ship you another. Parcels do sometimes go missing completely, but is very rare, less than 0.5%, but don't worry, I'll make sure you get your Guzunty.

If you would prefer an alternative course of action, please let me know, I'll be happy to accommodate you.

stevenverver, 2015-12-16:
Hi C
ampbellsan
 - it has arrived already! Thanks!