Guzunty/Pi #43: Guzunty no longer available on Amazon or EBay

gfischershaw, 2016-02-09:
Does not seem to be available any longer. Is Mark MacMillan going to be stocking them again, or should I be ordering direct?

Also, will higher capacity versions ever be made? I have a project (the wire interface for Lonmark over RS485, which ends up using 223 macrocells) currently running on a XC2C256-7-TQ144 on their CoolRunner II starter board. Of course a bit of a difference from what you currently have.

campbellsan, 2016-02-10:
Hi, sometimes Mark takes his page down temporarily if he's going on vacation or something.

However, this time may be more permanent I'm afraid.

Unfortunately Xilinx have EOL'd the PLC-44 package version of the XC9572XL device the Guzunty design uses. The smaller XC9536XL device which also fitted the board is also sadly discontinued. This means that it is rapidly getting more difficult and/or expensive to obtain the main part for the kit.

I still have a small stock of kits, but I'll be taking the purchase link down very soon.

 It's been a great fun designing and supporting the Guzunty but all things must come to an end sooner or later.

Or does it?

campbellsan, 2016-02-10:
I do have alternative designs that use a QFP version of the part and also offer a HAT compatible design with proper mounting points which could be rolled out as a Guzunty II.

However this design necessarily differs in its philosophy. The principle behind the Guzunty was that it could be easily assembled by someone even quite new to soldering and hacking. A QFP based design can be hand soldered, but I wouldn't recommend it for anyone new to the hobby. The design also loses the feature that a damaged device can't be easily swapped out if an experiment goes wrong.

Taking the second problem first, it turns out that I've only had reports of a tiny (~0.1%) percentage of devices failing once the build was complete, and only one report of a device getting damaged by a failed experiment. So perhaps this feature wasn't so important after all.

The first problem could perhaps be resolved by offering built boards (expensive compared to the cost of the pi onto which it fits), or kits with boards populated with just the device itself. 

This last idea leads to another thought ... if repopulated boards are an attractive option, why not populate it with a larger device? A Spartan 6 might be a very attractive option.

Let me know what you think.

gfischershaw, 2016-02-10:
Yes, sticking with a product when the EOL has been announced is not likely the best course of action. The very last thought on the Spartan 6 sounds quite attractive. Of course that is only my opinion. I had originally thought to try out the board on one of the new RPi Zero’s, as they appear to be completely compatible with an A+.