Guzunty/Pi #46: Ordered from US 8/30, status?

judemiller, 2016-09-16:
Sorry to use the issues list for this; couldn't figure out how to contact you. I ordered a loaded Guzunty Pi 8/30/16 and PayPal charged my credit card on your behalf.  Haven't heard anything yet and wondered if this is still a live site. I'm not in a hurry; just looking for some sign that I'm in the right place. Thanks,

George (Jude) Miller

campbellsan, 2016-09-16:
Hi George,

Yeah, I recall your Guzunty was held up a little waiting for some parts, but it shipped a while back. Should be with you soon.

I'll get back to you with the exact ship date when I get back to my desk.

It's rare for them to go missing, but on the occasion they do we'd ship you another.

Cheers,

Derek

judemiller, 2016-09-16:
No worries and thanks for the prompt response. I'm a retired hardware 
designer/low level, *ix driver guy and I think what you've done is 
fantastic. So sorry about the PLCC part being EOL.

Best,

Jude

judemiller, 2016-09-17:
Kit arrived safely in today's mail, thanks.

closed 2016-09-17 by judemiller