Guzunty/Pi #48: Guzunty @ Pi Zero

UseGitNotSvn, 2017-01-26:
Good evening everyone,

One thing first: An Guzunty and a Pi are a really enoyable combination, thx for making this work and also for the kits.

I got mine a few weeks ago, assembled it last week, but unfortunately I cannot get it work together with my Pi Zero (I dont have any other Pi at my hands). The soldering is well done, tribble-checked it already, but when I try to use Guzunty loader to get gz_test.xsvf on the CPLD nothing works.

### With 'sudo gz_load -v4 -p gz_test.xsvf' I get the following output:

Guzunty loader v5.01 rev 2, portions courtesy Xilinx, Inc.
Verbose level = 4
Progress Indicator suppressed
XSVF file = gz_test.xsvf
    sizeof( SXsvfInfo ) = 300 bytes
   TMS Reset Sequence -> Test-Logic-Reset
   TAP State = RESET
   XREPEAT = 32
   TMS Reset Sequence -> Test-Logic-Reset
   TAP State = RESET
   TAP State = RUNTEST/IDLE
   XRUNTEST = 0
   XSIR length = 8
   Shift Length = 8
    TDI          = 0xfe
    TDO Expected =
   TAP State = DRSELECT
   TAP State = IRSELECT
   TAP State = IRCAPTURE
   TAP State = IRSHIFT
   TAP State = IREXIT1
   TAP State = IRUPDATE
   TAP State = RUNTEST/IDLE
   XSDRSIZE = 32
    TDO Mask     = 0x0fffffff
   Shift Length = 32
    TDI          = 0x00000000
    TDO Expected = 0xf9604093
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
   TAP State = DRSELECT
   TAP State = DRCAPTURE
   TAP State = DRSHIFT
   TAP State = DREXIT1
   TAP State = DRUPDATE
   TAP State = RUNTEST/IDLE
 TDO Expected = 0xf9604093
 TDO Captured = 0x00000000
 TDO Mask     = 0x0fffffff

ERROR:  TDO mismatch and exceeded max retries
ERROR at or near XSVF command #8.  See line #8 in the XSVF ASCII file.

ERROR:  TDO mismatch and exceeded max retries
Execution Time = 0.052 seconds

My intuation tells me there is something wrong with gz_load trying to emulate the JTAG interface, because it gets no response from the CPLD. A quick test with a different development board (same CPLD though) lead to the same problem. According to Guzunty wiki it should work with Pi Zero. Any hints? I must have missed something? 

Regards

UseGitNotSvn, 2017-01-26:
After a little digging I fixed the problem. Had to adjust some lines of code in ports.c in gz_load sub directory. 

The problem was I am using a 'Raspberry Pi Zero Rev 1.2'. In setup_io() the device tree is read, and the model name of the raspberry Pi is extracted. The 2 within the model name lead to a problem with the current implementation: The software thinks a Pi 2 is used and sets peri_base to BCM2709_PERI_BASE, but BCM2708_PERI_BASE is the right selection for Pi Zero. 

After getting around of this, everything worked like a charm.

Cheers

campbellsan, 2017-01-28:
Hey great catch!

Could you please raise in a Github pull request? I will merge and publish the fix.

campbellsan, 2017-02-04:
Fix pushed. Tested on Pi 3 and Pi Zero (thanks UseGitNotSVN).

I had to forgo the printfs which show what peripheral set was detected because the output corrupts the progress indicator.

closed 2017-02-04 by campbellsan