Guzunty/Pi #50: Create a new Guzunty based on a surface mount device

campbellsan, 2017-05-08:


pefu, 2017-06-08:
Am I really the first to comment here?  I would be interested in buying a few.

garethjv, 2021-01-29:
I'm saddened there isn't more interest here. I guess it's the lack of publicity. I was fortunate to have grabbed one of these great little kits when they were still available some years ago and have to say they deserve much greater attention. They make great educational kits for VHDL development as well as offering all the I/O expansion possibilities for the Pi. Are there any other manufacturers of CPLDs that can be used instead of Xylinx?

campbellsan, 2021-01-31:
Hi Gareth,
Thanks! It sure is good to know that our little project was appreciated.

There are other CPLD manufacturers, but it's also worth noting that Xylinx did not discontinue the XC9500 series. They only discontinued the PC-44 package version. The Quad Flat Pack (QFP) version is still made for example. Basically, if a part won't fit in a mobile phone these days, any manufacturer needs a very compelling reason to keep making it.

The problem for the Guzunty specifically is that the PC-44 package was easy to assemble given relatively little soldering experience. By contrast, QFP's require a very steady hand and a supply of liquid flux. IOW, it's not really within the capability of many end users. We didn't want to bring to market a product with an inherent high return rate, that wouldn't be fun for anyone, so sadly we had to take the difficult decision to stop.

We did develop an Arduino clone HAT for the Pi featuring an XC9500XL for IO configuration (and ofc leveraging the 5v tolerance of the CPLD). It was based on the QFP package, but for the above reasons it never made it to market.

Looking to the future, the new RP2040 microcontroller looks interesting. Like the RPi, it is 3.3v only so who knows, maybe the Guzunty will be reborn in the near future?

BTW, check out the PIO state machine implementation in the RP2040. It may not be quite as flexible as a CPLD but it's capable of carrying out many of the speedy tasks the Guzunty was designed to handle in a $5, ready-to-use RPi foundation product.

cheers,
Derek

garethjv, 2021-02-01:
Hi Derek,
It was a nice surprise getting a reply to my comment. I agree totally with your response - SMD isn't for everyone. I know you can get QFP adapter boards for some devices but they aren't cheap.
Recently I've been playing around with FPGAs and actually bought a second hand DE1-SoC board at the end of last week which looks like it will be fun when I get time! But it was the Guzunty that got me hooked on wanting to learn about hardware description languages and I think it's a pity CPLDs aren't more readily available for hobbyists.
I hadn't heard of the RP2040 before - it looks interesting. I used to use PIC microcontrollers about 20 years ago for various things and the knowledge I learned with them was valuable for work when I was at a small start-up company and having multiple skills was valuable. I'll have to see how to get hold of one.
Thanks for all the info and all the best with your enterprises!
Gareth

garethjv, 2021-02-01:
I've just downloaded this month's MagPi and see there's lots about the Raspberry Pi Pico which uses the RP2040. That had completely passed me by. They're not in stock yet over here in the US so I'll have to wait to get hold of one. Thanks for the info!
Gareth